hudson-trading/slang-server #33: [Neovim] URI must contain a schema error

Steampunkery, 2025-09-25:
Hello,
Firstly, I'd like to thank everyone who works on this project for delivering what people like me have been wanting for a while but unable to get around to writing themselves: A good LSP implementation for SV using slang.

Secondly, I have received the below error a few times while using the server. It appears to come sporadically, and I'm not sure what exactly causes it. Sometimes I see it when I run `:LspRestart`, and sometimes it pops up while I'm actively editing.

```
Error executing vim.schedule lua callback: /opt/nvim-linux64/share/nvim/runtime/lua/vim/uri.lua:104: URI must contain a scheme:
stack traceback: 
    [C]: in function 'assert' /opt/nvim-linux64/share/nvim/runtime/lua/vim/uri.lua:104: in function 'uri_to_fname'
    ...im-linux64/share/nvim/runtime/lua/vim/lsp/diagnostic.lua:212: in function 'handle_diagnostics'
    ...im-linux64/share/nvim/runtime/lua/vim/lsp/diagnostic.lua:240: in function 'handler'
    /opt/nvim-linux64/share/nvim/runtime/lua/vim/lsp/client.lua:1116: in function '' vim/_editor.lua: in function <vim/_editor.lua:0>
```

toddstrader, 2025-09-25:
Thanks for reporting.  Looks like a borked diagnostic message:
https://microsoft.github.io/language-server-protocol/specifications/lsp/3.17/specification/#publishDiagnosticsParams
but it's hard to say what went wrong.

Can you capture the tail of `:LspLog` then next time you see this and post it here?

AndrewNolte, 2025-09-25:
I think this is likely due to our URI implementation, which can definitely be improved. It has a default constructor, which can make a URI with an empty string and therefore won't have a scheme. I think vscode just ignores these. I'll fix it to remove this possibility for diags at least. Longer term I'll improve the URI impl to be either FileUri or WebUri with no default constructor.

Steampunkery, 2025-09-27:
> Thanks for reporting. Looks like a borked diagnostic message: https://microsoft.github.io/language-server-protocol/specifications/lsp/3.17/specification/#publishDiagnosticsParams but it's hard to [...]

Here is the tail of `:LspLog` after it happened to me a few moments ago (paths redacted but otherwise unmodified):
```
[ERROR][2025-09-26 20:08:37] ...p/_transport.lua:36	"rpc"	"slang-server"	"stderr"	"<--- textDocument/didChange\nERROR: Token ';': Semicolon overlaps with previous token ']' : CloseBracket\n"
[ERROR][2025-09-26 20:08:37] ...p/_transport.lua:36	"rpc"	"slang-server"	"stderr"	"INFO: Analyzed /home/path/redacted/design.sv with tops: design\nERROR: Diagnostic has no ranges and no location: too many errors emitted, stopping now [--error-limit=]\n---> textDocument/publishDiagnostics\n---> textDocument/publishDiagnostics\nINFO: Published diags for /home/path/redacted/design.sv textDocument/didChange (notification finished)\n\n"
[ERROR][2025-09-26 20:08:41] ...p/_transport.lua:36	"rpc"	"slang-server"	"stderr"	"<--- textDocument/completion 130\nINFO: Completion triggered by: [' ','[']\nINFO: Falling back to sym kind for design.state_e.RESET\nINFO: Falling back to sym kind for design.state_e.SETUP\nINFO: Falling back to sym kind for design.state_e.SETUP1\nINFO: Falling back to sym kind for design.state_e.PROCESS\nINFO: Falling back to sym kind for design.state_e.DONE\n---> textDocument/completion 130\n\n"
[ERROR][2025-09-26 20:08:41] ...p/_transport.lua:36	"rpc"	"slang-server"	"stderr"	"<--- textDocument/didChange\nERROR: Token ';': Semicolon overlaps with previous token ']' : CloseBracket\n"
[ERROR][2025-09-26 20:08:41] ...p/_transport.lua:36	"rpc"	"slang-server"	"stderr"	"INFO: Analyzed /home/path/redacted/design.sv with tops: design\n---> textDocument/publishDiagnostics\nINFO: Published diags for /home/path/redacted/design.sv textDocument/didChange (notification finished)\n\n<-/- $/cancelRequest (ignoring threaded req)\n\n<--- textDocument/completion 131\nINFO: Completion triggered by: ['[',':']\nINFO: Falling back to sym kind for design.state_e.RESET\nINFO: Falling back to sym kind for design.state_e.SETUP\nINFO: Falling back to sym kind for design.state_e.SETUP1\nINFO: Falling back to sym kind for design.state_e.PROCESS\nINFO: Falling back to sym kind for design.state_e.DONE\n---> textDocument/completion 131\n\n"
[ERROR][2025-09-26 20:08:41] ...p/_transport.lua:36	"rpc"	"slang-server"	"stderr"	"<--- completionItem/resolve 132\n---> completionItem/resolve 132\n\n"
```

Steampunkery, 2025-10-01:
Oh I thought I already replied on here. Yeah it totally solved it, thanks!